redpomodoro/fronius_modbus #1: Fronius Modbus Integration Fails to Load: 'No setup or config entry setup function defined' Error

Klangen82, 2024-12-05:
**Describe the issue you are experiencing**
I am unable to add the inverter using the Fronius Modbus integration.

**Describe the expected behavior** 
The inverter should be added as a device successfully.

**To Reproduce** 

1. Downloaded the Fronius Modbus integration and placed it under \\IP-ADDRESS\config\custom_components\fronius_modbus.
2. Added the integration via + Add Integration.
3. Entered the inverter's IP address and the address (200 in my case).
4. Pressed Submit. The process appeared successful, but the integration failed to load. (even after reboot)

[image] 

I looked in the log and see the following error message: 
[image] 

I use Node-Red with the mobus settings and it's working for me, can control battery etc. I have disabled the Node-Red Mobus before I tried to add this integration just to be sure there where no clonflicts. 

**What version of Home Assistant Core is installed** 
Core 2024.12.0
Supervisor 2024.11.4
Operating System 14.0
Frontend 20241127.4 

**What version of Fronius Modbus integration is installed** 
Release 0.1.0 

**What Inverter and extras are you using** 
Gen24: 8kW (Firmware 1.33.7-1) 
SmartMeter: TS 65A-3 
Battery System: BYD Battery-Box Premium HV 

**Modbus setup** 
[image] 

**Error message shown within Home Assistant log** 
Logger: homeassistant.setup Source: setup.py:269 First occurred: 17:21:17 (1 occurrences) Last logged: 17:21:17 Setup failed for custom integration 'fronius_modbus': No setup or config entry setup function defined. 

**Diagnostics information of Fronius Modbus integration** 
As the integration doesn’t load, diagnostics information is not available.

redpomodoro, 2024-12-07:
I assume it has to do with that you use SunSpec Model Type 'float' while the integration is using the default 'int +SF'. I'll add this to the documentation as note and see if the integration and catch this in the config flow.

[image]

Rainman67, 2024-12-07:
@redpomodoro : I have the same problem as @Klangen82, but I tried it with setting "int + SF", but I get still the same error message as before with the setting "float". Is there another possible solution?

Sorry, but my English is not the best! ;-)

redpomodoro, 2024-12-08:
@Klangen82 and @Rainman67 
Just noticed that I forget to add the __init__.py to the GitHub repository. When this file is missing the setup will fail. Can you try copying this file into your local folder too?

Rainman67, 2024-12-08:
@redpomodoro : now I could configure the integration! Thanks for your tip! But unfortunately, no values ​​are currently displayed, probably because the inverter is in sleep mode. Can I only control with your integration if the inverter is not in sleep mode?

redpomodoro, 2024-12-09:
Great that you now can setup the integration!

My inverter doesn't sleep since it has been connected to the battery system. But it would make sense that it sleeps it won't allow and changes.  Maybe I can add check to see if the inverter is sleeping.

I've started to add some description on the data points and what the actions do:

https://github.com/redpomodoro/fronius_modbus/blob/main/README.md#usage

To make it clearer I will add another table to show which setting is being used by the various actions.

Klangen82, 2024-12-09:
I got it working as well but no sensors first then I change the inverter to be in "int + SF" then I got some sensors working. But most of them is not working and 

[image]
[image]
[image]

```
2024-12-09 10:24:39.567 INFO (SyncWorker_0) [custom_components.fronius_modbus.hub] read registers a: 40071 s: 200 c 103
2024-12-09 10:24:39.580 INFO (SyncWorker_0) [custom_components.fronius_modbus.hub] read registers a: 40255 s: 200 c 88
2024-12-09 10:24:39.587 ERROR (SyncWorker_0) [custom_components.fronius_modbus.hub] modbus mppt error Exception Response(131, 3, IllegalAddress)
2024-12-09 10:24:39.588 INFO (SyncWorker_0) [custom_components.fronius_modbus.hub] read registers a: 40345 s: 200 c 24
2024-12-09 10:24:39.601 ERROR (SyncWorker_0) [custom_components.fronius_modbus.hub] modbus storage error Exception Response(131, 3, IllegalAddress)
2024-12-09 10:24:39.601 INFO (SyncWorker_0) [custom_components.fronius_modbus.hub] read registers a: 40071 s: 200 c 38
2024-12-09 10:24:39.613 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/fronius_modbus/hub.py", line 125, in async_refresh_modbus_data
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/fronius_modbus/sensor.py", line 151, in _modbus_data_updated
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1038, in _stringify_state
    return str(state)
```

[image]

redpomodoro, 2024-12-11:
It seems for some reason the inverter device was not setup correctly and it seems read the from the smart meter. I've added more checks into the code during the configuration process. So you can try removing the integration, update the code to latest version, restart HA and then add the integration again.

Klangen82, 2024-12-11:
Got all sensors to work:
[image]
[image]
[image]

If I change "Storage Control Actions" I get an error "Failed to perform the action select/select_option. 'Hub' object has no attribute '_device_address'"

Rainman67, 2024-12-11:
Thank you very much for the effort and work you put into this project!

I have just imported the changed files and tested the integration again. As it stands, it works perfectly with one exception. The “Storage Control Actions” item also works as it should.

However, if I charge my battery from the grid, the “Grid Charge Power” controller does not work as expected. It always charges at 6 kW, no matter what I set. What is the reason for this?

And a question about the “Reserve Target” function: what does this menu item control?

redpomodoro, 2024-12-12:
@Klangen82  that bug should be resolved

@Rainman67 I'm working on new version that makes controlling the storage easier to understand. In the current form changing settings needs to follow a certain order work.

Rainman67, 2024-12-12:
@redpomodoro : thank you! I will wait for the new release!

Rainman67, 2024-12-15:
@redpomodoro : Hi, I saw that you redid your integration. Of course I had to test it immediately and now, unlike before, everything works perfectly. Thanks for that! I adapted your integration a little for myself. On the one hand, I translated it into German, and on the other hand, I added two additional charging modes (min. SoC 20% and min. SoC 70% for the emergency power reserve in winter). Based on this, I created an automation that changes the charging mode to "SoC 20%" after the SoC reaches 100%. In addition, there are two automations that control the SoC depending on the solar forecast (if little comes from the roof, then 70% SoC is set, if a lot comes the next day, then SoC 20% is set). I also created a script that calibrates the battery. To do this, the battery is first fully charged, then discharged to 5 percent and then fully charged again. After this, finally the SoC is automatically set to 70% again. I hope it works that way.

At this point, thank you again for the time and work you put into this project!

vistalba, 2024-12-15:
The installation with HACS doesn’t work. 
I assume it is because the „release“ ist already 3 weeks old and you added the init.py file just some days ago.

Could you please push a new „release“?

redpomodoro, 2024-12-16:
> The installation with HACS doesn’t work. I assume it is because the „release“ ist already 3 weeks old and you added the init.py file just some days ago.
> 
> Could you please push a new „release“?

Can you see it in HACS? As far as I'm aware the integration is still pending to be included into HACS it seems there is quite a backlog of requests to add additional integrations, see:

https://github.com/hacs/default/pull/2872

So the only option is to manually copy the files into your custom_components folder.

redpomodoro, 2024-12-16:
> @redpomodoro : Hi, I saw that you redid your integration. Of course I had to test it immediately and now, unlike before, everything works perfectly. Thanks for that! I adapted your integration a [...]

Great to hear it works as expected now! I don't have my battery that long, but in your experience how often the battery needs to be calibrated? Might be a good feature to add to the integration.

vistalba, 2024-12-16:
> > The installation with HACS doesn’t work. I assume it is because the „release“ ist already 3 weeks old and you added the init.py file just some days ago.
> > Could you please push a new „release“?
> 
> Can you see it in HACS? As far as I'm aware the integration is still pending to be included into HACS it seems there is quite a backlog of requests to add additional integrations, see:
> 
> [hacs/default#2872](https://github.com/hacs/default/pull/2872)
> 
> So the only option is to manually copy the files into your custom_components folder.

You can add a custom repository like described here: https://www.hacs.xyz/docs/faq/custom_repositories/

Installation does work but just with the version which is a "release". (Not able to install "main" branch with HACS).

Rainman67, 2024-12-16:
@redpomodoro :

I have a BYD battery with 10.2 kWh (LiFePO4 technology). According to experts, batteries with LiFePO4 technology in particular need to be fully charged more often in order to carry out the calibration. When I fully charged my battery a few days ago, I noticed that the battery was charged relatively quickly to 99.2%. But then it took almost 50 minutes until 100% was reached. During this time, the battery was charged with 6 kW for 40 minutes. The last 10 minutes it was 3 kW. Previously the battery lasted for about 10 days at 70% displayed capacity (Emergency power reserve in winter). 

My suspicion is confirmed by the fact that I ran the calibration yesterday once again. The battery was then first charged again to 100%. Then discharged to the minimum of 5% and fully charged again to 100%. By the way, when I last fully charged it, it only took 2 minutes from 99.2% to 100%, which suggests that the battery is now actually calibrated. That's why I'll start my script in the Home Assistant regularly once a week from now on, as long as the battery isn't being charged by the sun.

vistalba, 2024-12-17:
Upgrading via HACS to 0.1.1 worked well for me. Thx!

redpomodoro, 2024-12-18:
> @redpomodoro :
> 
> I have a BYD battery with 10.2 kWh (LiFePO4 technology). According to experts, batteries with LiFePO4 technology in particular need to be fully charged more often in order to carry out the calibration. When I fully charged my battery a few days ago, I noticed that the battery was charged relatively quickly to 99.2%. But then it took almost 50 minutes until 100% was reached. During this time, the battery was charged with 6 kW for 40 minutes (!!). The last 10 minutes it was 3 kW. Previously the battery lasted for about 10 days at 70% displayed capacity (Emergency power reserve in winter).
> 
> My suspicion is confirmed by the fact that I ran the calibration yesterday once again. The battery was then first charged again to 100%. Then discharged to the minimum of 5% and fully charged again to 100%. By the way, when I last fully charged it, it only took 2 minutes from 99.2% to 100%, which suggests that the battery is now actually calibrated. That's why I'll start my script in the Home Assistant regularly once a week from now on, as long as the battery isn't being charged by the sun.
> 
> EDIT: I have another question in this context: there is a tool with which you can read a lot of data from BYD batteries via Modbus (BYD logger from Olli). Apparently you can also query the battery health (SOH). Can you please tell me how I can get this into your integration?

I've created a new request for additional data points, see: https://github.com/redpomodoro/fronius_modbus/issues/5

redpomodoro, 2024-12-18:
> @redpomodoro :
>
> I have a BYD battery with 10.2 kWh (LiFePO4 technology). According to experts, batteries with LiFePO4 technology in particular need to be fully charged more often in order to carry [...]

Concerning the calibration I've added also another feature request for this: 
https://github.com/redpomodoro/fronius_modbus/issues/6

redpomodoro, 2024-12-18:
> Upgrading via HACS to 0.1.1 worked well for me. Thx!

Good to know I'll setup my own production HA server to use this way as well then I can also check if future releases are pulling through

redpomodoro, 2024-12-18:
As the original issue has been solved, closing this one.

closed 2024-12-18 by redpomodoro