redpomodoro/fronius_modbus #6: Feature: Calibration option

redpomodoro, 2024-12-18:
As suggested by @Rainmain67 it will be good to calibrate the battery time to time, by fully charging and discharging it.

redpomodoro, 2024-12-18:
I'm testing to see if I can add an option for this

vistalba, 2024-12-18:
I have implemented this automation using NodeRed and the respective integration.

Here is how the automation works:

- When the battery charge reaches 92%, the charging rate is reduced to 12% (equivalent to approximately 1.5kW).
- When the battery charge reaches 97%, the charging rate is further reduced to 3% (around 385W), and the discharge limit is set to 0%. At this point, no further discharging is possible unless the "Auto" setting is set later.
- When the battery charge reaches 99%, the charging rate is lowered to 1% (around 128W).
- When the battery charge reaches 100%, a 60-minute waiting period is initiated before setting it back to "Auto" mode. This gives the balancer 60min to balance the cells.

I have identified a potential issue with this automation. When the battery charge reaches 97% and the sun goes down, the battery can't be charged anymore due to the lack of PV power. Additionally, it can't be discharged due to the 0% discharge limit. As a result, it never reaches 100% and doesn't switch back to "Auto" mode.

To address this, I plan to implement a detection mechanism that identifies this situation and allows grid charging until the battery reaches 100%. A 60-minute balancing period will follow before switching back to "Auto" mode.

I would appreciate any thoughts or suggestions on improving this "calibration" process. Please note that this automation is still in the early alpha stage, as I just developed it last weekend.