redpomodoro/fronius_modbus #9: Beginner question, auto don´t work after charging from grid

MickeSB, 2025-01-01:
Hi, 

I have just started testing the integration and have found a problem I don´t understand. 

When using charging from grid everything works as expected and I can control the level of charging. But after returning to auto and using my usual inverter controlled night charging the battery is stuck on "holding" and will not discharge to the grid any more.

Is there something I must do to return to full automation again, where the inverter controls charging and discharging?

Regards
//Mikael
[image]

redpomodoro, 2025-01-05:
Just to understand you use the scheduled nightly charging in the Fronius WebUI to charge it?

MickeSB, 2025-01-05:
Not any more, I use your integration for both charging and returning to Auto. I have found a workaround for my problem. I´ve made an automation where I change from charging to discharging and after 30sec go to Auto. So for the moment my automations works and I think after testing that the problem with going to Auto is intermittent, sometimes it works and sometimes not, so the problem can be in the communication and not in the integration. But if you or someone has an idea how to pinpoint the problem I´m very interested. Otherwise the integration is great and does everything I need.

redpomodoro, 2025-01-14:
Thanks for the additional info. I'll will do some more testing to see if I can reproduce the problem.

redpomodoro, 2025-01-20:
I have not been able to reproduce it, but however I noticed that the Charge status seems to some threshold to change status. It can remain in holding as long as the charge/discharge currents are small. So if your home uses little energy after being charged at night, it could take until more power usage before it will change to discharging.

redpomodoro, 2025-03-03:
Please reopen if the issue this occurs

closed 2025-03-03 by redpomodoro