redpomodoro/fronius_modbus #10: OverflowError: int too large to convert to float

vistalba, 2025-01-04:
**Describe the issue you are experiencing**
After upgrade to 2025.1.0 I got errors in the HA logs.

**What version of Home Assistant Core is installed**
2025.1.0

**What version of Fronius Modbus integration is installed**
v0.1.2

**What Inverter and extras are you using**
Gen24: 10kW 
SmartMeter: yes
Battery System: BYD HVM

**Error message shown within Home Assistant log**
```
Logger: custom_components.fronius_modbus.hub
Quelle: custom_components/fronius_modbus/hub.py:243
Integration: Fronius Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 12:41:07 (5 Vorkommnisse)
Zuletzt protokolliert: 12:42:07

Error reading inverter storage data
Traceback (most recent call last):
  File "/config/custom_components/fronius_modbus/hub.py", line 243, in _refresh_modbus_data
    update_result = self.read_inverter_storage_data()
  File "/config/custom_components/fronius_modbus/hub.py", line 654, in read_inverter_storage_data
    self.data['load'] = self.data['m1_power'] + self.data['acpower']
                        ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
OverflowError: int too large to convert to float
```

vistalba, 2025-01-10:
I think the issue comes from the Fronius Smart Meter:
[image]

Any idea, why this happens and how I could fix or troubleshoot this?

redpomodoro, 2025-01-14:
There was a mistake in fetching the AC_Power in 0.1.2/0.1.3 releases. Addressed that in the upcoming 0.1.4 release and it should fix the error in that is shown in the logs. However that doesn't explain why the exported number is off.

redpomodoro, 2025-01-20:
@vistalba is this still occurring?

vistalba, 2025-01-20:
Yes, this is still an issue with v 0.1.4. Two log entries:

```
Logger: custom_components.fronius_modbus.hub
Quelle: custom_components/fronius_modbus/hub.py:253
Integration: Fronius Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 17:49:42 (4 Vorkommnisse)
Zuletzt protokolliert: 17:50:12

Error reading meter data 1.
Traceback (most recent call last):
  File "/config/custom_components/fronius_modbus/hub.py", line 253, in _refresh_modbus_data
    update_result = self.read_meter_data(meter_prefix="m1_", unit_id=meter_address)
  File "/config/custom_components/fronius_modbus/hub.py", line 702, in read_meter_data
    self.data['load'] = acpower + inverter_acpower
                        ~~~~~~~~^~~~~~~~~~~~~~~~~~
OverflowError: int too large to convert to float
```

```
Logger: homeassistant.helpers.entity
Quelle: helpers/entity.py:1227
Erstmals aufgetreten: 17:49:42 (4 Vorkommnisse)
Zuletzt protokolliert: 17:50:12

Failed to set state for sensor.meter_1_power, fall back to unknown
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1227, in _async_write_ha_state
    hass.states.async_set_internal(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<5 lines>...
        time_now,
        ^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2364, in async_set_internal
    state = State(
        entity_id,
    ...<8 lines>...
        timestamp,
    )
  File "/usr/src/homeassistant/homeassistant/core.py", line 1808, in __init__
    validate_state(state)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 237, in validate_state
    raise InvalidStateError(
    ...<2 lines>...
    )
homeassistant.exceptions.InvalidStateError: Invalid state with length 1005. State max length is 255 characters.
```

Then there are more modbus related log entries, but I don't know if it is related:
``` 
Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 17:49:29 (1 Vorkommnisse)
Zuletzt protokolliert: 17:49:29

Setup failed for 'modbus': Unable to import component: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
[... 36 lines cut ...]
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/modbus/__init__.py", line 142, in <module>
    from .modbus import ModbusHub, async_modbus_setup
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 17, in <module>
    from pymodbus.framer import FramerType
ImportError: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)
```

```
Logger: homeassistant.util.loop
Quelle: util/loop.py:77
Erstmals aufgetreten: 17:49:29 (1 Vorkommnisse)
Zuletzt protokolliert: 17:49:29

Detected blocking call to import_module with args ('homeassistant.components.modbus',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)
```

redpomodoro, 2025-01-21:
For the first 2 errors I've put in some extra error handling that should handle the issue more gracefully, however I still don't get the underlying issue. Hopefully the logs will provide some more clues with the new error handling.

The last error is related to the modbus integration, which seems to expect a newer version of pymodbus library.

redpomodoro, 2025-01-22:
Released in v0.1.5. Please let me know if continues to occur and share your logs so that we can pinpoint the issue.

vistalba, 2025-01-25:
I still see this issue with 0.1.5 :-/

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.fronius_modbus.hub
Quelle: custom_components/fronius_modbus/hub.py:271
Integration: Fronius Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 15:25:54 (56 Vorkommnisse)
Zuletzt protokolliert: 15:39:04

Error reading meter data 1.
Traceback (most recent call last):
  File "/config/custom_components/fronius_modbus/hub.py", line 271, in _refresh_modbus_data
    update_result = self.read_meter_data(meter_prefix="m1_", unit_id=meter_address)
  File "/config/custom_components/fronius_modbus/hub.py", line 727, in read_meter_data
    self.data['load'] = round(acpower + inverter_acpower,2)
                              ~~~~~~~~^~~~~~~~~~~~~~~~~~
OverflowError: int too large to convert to float
```

```
Logger: homeassistant.helpers.entity
Quelle: helpers/entity.py:1227
Erstmals aufgetreten: 15:25:54 (157 Vorkommnisse)
Zuletzt protokolliert: 15:39:04

Failed to set state for sensor.meter_1_power, fall back to unknown
Failed to set state for sensor.load, fall back to unknown
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1227, in _async_write_ha_state
    hass.states.async_set_internal(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<5 lines>...
        time_now,
        ^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2364, in async_set_internal
    state = State(
        entity_id,
    ...<8 lines>...
        timestamp,
    )
  File "/usr/src/homeassistant/homeassistant/core.py", line 1808, in __init__
    validate_state(state)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 237, in validate_state
    raise InvalidStateError(
    ...<2 lines>...
    )
homeassistant.exceptions.InvalidStateError: Invalid state with length 1005. State max length is 255 characters.
```

Any further logs I can grab to get more detailed infos?

redpomodoro, 2025-01-26:
Does it report the meter model type and firmware correctly in the Device Info section? 

I wonder if a different meter type might report the data differently by using a different data structure. I'm using a TS65A-3.

vistalba, 2025-01-27:
When I check it over the Fronius integration (API, not Modbus) it is reported as: Smart Meter TS 65A-3

However, the Fronius Modbus integration shows it like this and the device name is "Fronius Meter 1":

[image]

redpomodoro, 2025-01-27:
The device info (geraete-info) should display the firmware number and model from the meter and not the inverter. So it seems it already fails at getting the base data. 

You could try removing the integration completely and add it again. 

If that doesn't work note that I already started adding more checks when the meter is initialized, so hopefully that gives more clues on in case it fails to get the meter data.

vistalba, 2025-01-27:
Deleted integration, reinstalled v0.1.5, restarted HA and then started from scratch... seems to have fixed this issue. Strange. Sorry about that :(

[image]

closed 2025-01-27 by vistalba