redpomodoro/fronius_modbus #11: homeassistant.exceptions.InvalidStateError: Invalid state with length 989. State max length is 255 characters.

vistalba, 2025-01-04:
**Describe the issue you are experiencing**
After upgrading HA to 2025.1.0 I noticed this error in HA logs.

**What version of Home Assistant Core is installed**
2025.1.0

**What version of Fronius Modbus integration is installed**
v0.1.2

**What Inverter and extras are you using**
Gen24: 10kW
SmartMeter: yes
Battery System: BYD HVM

**Error message shown within Home Assistant log**
```
Logger: homeassistant.helpers.entity
Quelle: helpers/entity.py:1227
Erstmals aufgetreten: 12:40:57 (16 Vorkommnisse)
Zuletzt protokolliert: 12:44:17

Failed to set state for sensor.meter_1_power, fall back to unknown
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1227, in _async_write_ha_state
    hass.states.async_set_internal(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<5 lines>...
        time_now,
        ^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2364, in async_set_internal
    state = State(
        entity_id,
    ...<8 lines>...
        timestamp,
    )
  File "/usr/src/homeassistant/homeassistant/core.py", line 1808, in __init__
    validate_state(state)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 237, in validate_state
    raise InvalidStateError(
    ...<2 lines>...
    )
homeassistant.exceptions.InvalidStateError: Invalid state with length 989. State max length is 255 characters.
```

redpomodoro, 2025-01-05:
Tested new dev version 0.1.3 with HA 2025.1.0 and it seems to work fine.

redpomodoro, 2025-01-14:
Great!

closed 2025-01-14 by redpomodoro

vistalba, 2025-01-14:
Updated, still have this issue with my power meter:
```
Logger: homeassistant.helpers.entity
Quelle: helpers/entity.py:1227
Erstmals aufgetreten: 15:40:21 (769 Vorkommnisse)
Zuletzt protokolliert: 17:53:32

Failed to set state for sensor.meter_1_power, fall back to unknown
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1227, in _async_write_ha_state
    hass.states.async_set_internal(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        entity_id,
        ^^^^^^^^^^
    ...<5 lines>...
        time_now,
        ^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2364, in async_set_internal
    state = State(
        entity_id,
    ...<8 lines>...
        timestamp,
    )
  File "/usr/src/homeassistant/homeassistant/core.py", line 1808, in __init__
    validate_state(state)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 237, in validate_state
    raise InvalidStateError(
    ...<2 lines>...
    )
homeassistant.exceptions.InvalidStateError: Invalid state with length 1004. State max length is 255 characters.

```