redpomodoro/fronius_modbus #13: Feature request: Show limits in Watts instead of %

redpomodoro, 2025-01-06:
It would be great to have the charge/discharge limits in W or kW rather than a % of the total power. Not sure if this can be done automatically at the time of setting up the device once the maximum capacity has been read, or if it would require the user to provide it beforehand in order to set the limits to the sliders...

As suggested by:
https://community.home-assistant.io/t/fronius-modbus-integration/803051/18

vistalba, 2025-01-07:
I already thought about this too. But its a bit complicated.

The current percent implementation 0-100% is based on the 12800W limit of the inverter. So we are talking about 128W per %.
This limit is only the absolut max. which is possible.
Depending on Inverter, Battery and amount of Battery Modules combinatin, the real limit is different.

For my example:

Fronius Gen24 10kw: 22A limit
Battery BYD HVM: max. output is 50A / Nominal Voltage dependes on amount of Modules.

So for me the limit with BYD HVM 5 Modules and Gen24 combination the limit is always 22A*Battery-Voltage. So the max. in W is around 6000W. 

How should we now calculate back to the 0-100% logic of Fronius? As 128W * 47 is already 6kW. Only 0-47% are real values for my case. If I would buy additional Battery Modules the range gets bigger.

Any idea how this could be solved?

redpomodoro, 2025-01-10:
@vistalba I don't think it is problem for the limits, but only when you do forced charging or discharging were the actual numbers will be lower like you said. This could be overcome by adding a configuration setting to integration allowing the user could set the real maximum forced charge / discharge rate or by calculating the maximum based on the configuration. 

I will first change the controls to watts and then we see how much a problem this is actually is.

redpomodoro, 2025-01-10:
This planned to go out in the 0.1.4 release

nicolaufg, 2025-01-19:
Got 0.1.4 and it looks great!
Thanks redpomodoro!

redpomodoro, 2025-01-20:
Cool! 

Closing this one.

closed 2025-01-20 by redpomodoro