redpomodoro/fronius_modbus #16: Feature request: offgrid or mode sensor

tompaah, 2025-01-06:
It would be nice if the integration could provide a sensor that tells if the Gen24 inverter is in on- or offgrid operation. If there is a modbus register in the inverter for its operational mode, please add it to the integration. This would enable switching off unnecessary loads during power outages. Currently the official Fronius integration does not provide any sensor telling if the inverter i grid-connected or not.

My Gen24 is configured for Full Backup and if you need any testing I can do this for you.

Again, many thanks for the integration already in its current state!

redpomodoro, 2025-01-15:
I noticed that the ECP just provide the status of the electrical connection at the inverter and this will turn off when the inverter is sleeping. I've added voltage at the PCC so that could be an indicator if that drops. As well several other status fields.

tompaah, 2025-01-15:
Okay, I'll test this when the 0.1.4 is released.

redpomodoro, 2025-01-17:
Turns out the PCC voltage is just a fixed reference value, so that won't be helpful. My guess is that monitoring the line-to-line voltage of the grid point meter would work best. 

However I've added more diagnostic fields that might get triggered when the grid connection is lost.

tompaah, 2025-01-18:
Tested a power outage today.

`sensor.fronius_inverter_symo_gen24_10_0_events` changes from `none` to `info` in offgrid mode but I assume this is not a reliable indicator since info events could be almost anything, not related to offgrid operation.

`sensor.fronius_inverter_symo_gen24_10_0_line_frequency` displays ~ `50.xx` Hz in ongrid operation but goes to `53.00` Hz in offgrid mode. This is expected since the inverter runs at 53 Hz in offgrid mode according to the manual. However this value is at `0.00` Hz during inverter mode changes, which take up to 2 minutes to complete (inverter startup after power outage). This might be the most reliable indicator right now, to deem if the system is operating in on- or offgrid mode.

No other sensors in the `sensor.fronius_inverter_symo_gen24_10_0*` group changed values significantly during the test.

tompaah, 2025-01-19:
Reading a [german forum thread](https://www.photovoltaikforum.com/thread/183588-fronius-modbus-notstrom-erkennung/) on the topic, it seems indeed the line frequency value is the most reliable indicator. There appears not to be any other specific registry value for indicating wether the inverter is in offgrid mode or not.
I'll close this issue as my conclusion is that the line frequency is what there is and it is already featured in the integration.

closed 2025-01-19 by tompaah

redpomodoro, 2025-01-20:
Thanks for testing! Good to know that the inverter goes to 53hz when offgrid. I wonder if that is the same when the PV Point is used. I'll test that later and maybe add a status data point which checks the frequency.

vistalba, 2025-01-25:
> Thanks for testing! Good to know that the inverter goes to 53hz when offgrid. I wonder if that is the same when the PV Point is used. I'll test that later and maybe add a status data point which checks the frequency.

Be aware that this Frequency can be configured in the Gen24 inverter! 

For me it is configured to 51.6Hz. (Menu on German: Gerätekonfiguration -> Wechselrichter -> Notstrom)

[image]