redpomodoro/fronius_modbus #17: Upgrade from 0.1.3 to 0.1.4 made some battery controls disabled

jorgen-zafir, 2025-01-17:
Upgraded to version 0.1.4 from 0.1.3 and most battery control became unavailable. Reverting to version 0.1.3 resolved the issue. I didn't record exactly what entities that were affected but looking at the log it seems to be:

- Discharge Limit
- Charge Status
- Storage Control Mode
- Grid Charging

I didn't check any other devices as their corresponding entities are available from the Fronius integration, which connects to the SolarNet REST API.

Error messages emitted for v0.1.4:
``2025-01-17 14:59:55.146 ERROR (SyncWorker_3) [custom_components.fronius_modbus.hub] Error storage json data http://gen24.intra.greytower.net/solar_api/v1/GetStorageRealtimeData.cgi '1'
Traceback (most recent call last):
  File "/config/custom_components/fronius_modbus/hub.py", line 209, in get_json_storage_info
    details = data['Body']['Data']['1']['Controller']['Details']
              ~~~~~~~~~~~~~~~~~~~~^^^^^
KeyError: '1'

But the same error message is emitted by the 0.1.3 version so I don't think it's relevant. The _GetStorageRealtimeData.cgi_ script do actually work when pointing a web browser towards it.

I don't have any test environment so I will not be that helpful in testing any patches.

redpomodoro, 2025-01-17:
It looks like it could not retrieve the name and model details for the battery storage. I made the code more robust to continue if the name/model cannot be found. The fix will go out in the next release v0.1.5 release.

redpomodoro, 2025-01-22:
Released in v0.1.5. Hopefully this resolves the issue.

redpomodoro, 2025-03-03:
Please reopen if the issue still is occuring.

closed 2025-03-03 by redpomodoro