redpomodoro/fronius_modbus #19: what is "load"?

jaegerschnitzel, 2025-01-21:
First of all I'm using the addon only for getting data out of my inverter. I don't have a battery. 
But I'm not sure what the "Load" sensor under Inverter is telling me. Unfortunately there is no description that's why I'm asking here.
The only thing I found is this code, but I cannot handle it ;-)

```
        if meter_prefix == 'm1_':
            inverter_acpower = self.data.get('acpower')
            if not acpower is None and not inverter_acpower is None:
                self.data['load'] = acpower + inverter_acpower
```

Would be very nice if you can explain what load is because the numbers are really strange and I cannot interpret it.

redpomodoro, 2025-01-27:
The load is the current total power consumption which is derived by adding up the meter AC power and interver AC power.

It seems that you experience the same issue as reported here: https://github.com/redpomodoro/fronius_modbus/issues/10

jaegerschnitzel, 2025-01-27:
Thanks for the explanation. Yes it was the same problem and linked. And after readding the integration it's fixed.

closed 2025-02-06 by jaegerschnitzel