redpomodoro/fronius_modbus #20: Make offgrid sensor aware of offset setting

redpomodoro, 2025-01-27:
Follow up from: https://github.com/redpomodoro/fronius_modbus/issues/16#issuecomment-2613933922

The current offgrid state is based on the default offset of 3hz, but it can be configured differently. It would be better if the integration can use the offset value instead.

redpomodoro, 2025-02-10:
Implemented a smaller range to change deviation from 50hz. I could not find a way to read to configured offset value through modbus.

vistalba, 2025-02-10:
> Implemented a smaller range to change deviation from 50hz. I could not find a way to read to configured offset value through modbus.

May provide input_number or value on installation could be set to define custom offset value?

redpomodoro, 2025-02-12:
The minimum offset is 0.5hz I recall. I defined on grid as 50hz +/- 0.2hz so it should work for any configured value.

redpomodoro, 2025-03-03:
Please reopen if the issue still is occuring.

closed 2025-03-03 by redpomodoro