redpomodoro/fronius_modbus #22: Varoious modbus issues

Casca80, 2025-02-07:
I just tried the integration (running on 2025.2.0) resulting in a semi working instance. Meter, Inverter works, about half of the battery sensors / controls.

Seems to be modbus related. Did double check int-SF / float settings on inverter. 
Have other modbus components that also stopped working.

[Logger homeassistant.setup.txt](https://github.com/user-attachments/files/18705969/Logger.homeassistant.setup.txt)

[image]

redpomodoro, 2025-02-10:
The errors in the log are related to modbus binding and not to this binding. Also the data is populated. 

Note that only the controls are enabled that are relevant for the current control mode. See the documentation for mapping:

https://github.com/redpomodoro/fronius_modbus?tab=readme-ov-file#controls-used-by-modes

stevenvandenbrandenstift, 2025-03-02:
From what i read here and tested myself, 
The issue seems to be that integrations share the same python packages and thus only 1 pymodbus version can remain. The Internal modbus Integration of ha is upgraded to a New version, while this Integration is still on an older One. 

https://github.com/home-assistant/core/issues/134593

I verified this by removing this Integration ( and removing my modbus configuration.yaml ) , rebooting ha and putting the modbus config back. Then modbus Works again.

redpomodoro, 2025-03-03:
Good to know. I'll add this to documentation to help other users and this version conflict might happen in the future.

redpomodoro, 2025-03-03:
Closing as fix has been provided by [stevenvandenbrandenstift](https://github.com/stevenvandenbrandenstift)

stevenvandenbrandenstift, 2025-03-03:
I guess the only possible way is attempting to stay in sync with the ha version, but the design seems hard to maintain.