redpomodoro/fronius_modbus #23: Charge from Grid limit >500W not working

vistalba, 2025-02-10:
I don't know if there is some issue with my battery... but I can't get a "charge from grid" limit above 500W to work.
I can set whatever I want above 500W, but it doesn't charge faster.
Limits below 500W are working fine (e.g. 205W)

I don't know if there is an issue since the change from % to W. As I remember, it has already worked to charge faster.

Edit:
I see that only 10W steps are allowed but also 1W are accepted.
Charging with PV energy does work fast (up to 9kW - 22A limit of the Inverter) so I assume it is not a battery issue.

redpomodoro, 2025-02-10:
Unfortunately I can't reproduce the error. Are you putting in integer values without any separators? i.e. 2000 should work, but 2,000 won't work.

redpomodoro, 2025-02-10:
Another thing to check if you put in say 2000 watt and then check 'discharging power'. In my case it is almost 10% for a 10kW inverter.

wubb84, 2025-03-02:
You have to update your inverter, and the problem is gone

vistalba, 2025-03-02:
> You have to update your inverter, and the problem is gone

Which version do you have which is working?

wubb84, 2025-03-02:
I don’t know which inverter you have. I have a Symo Gen24. I have the same problem before the update at lasted version now. You have read the changelog of updates by your inverter.

redpomodoro, 2025-03-03:
My inverter has version 1.34.6-1 which runs fine. But I noticed there is newer version 1.35.4-1 is available.

wubb84, 2025-03-03:
1.35.4 solved this Problem. Read Changelog. 
“ Bugfixes
/ Verbesserungen in der Weboberfläche bei Updatevorgang implementiert.
/ Fehler bei der Berechnung und Darstellung im Web UI für asymmetrische Einspeisung behoben.
/ Die Funktion ”Batterieladung aus anderen Generatoren im Heimnetzwerk” funktionierte manchmal nicht.
/ State 1197 Fehlauslösungen behoben.
/ Durch Änderungen in der RCMU Signalaufbereitung kam es vereinzelt zu Fehlauslösungen. Diese Änderungen wurden wieder
zurückgenommen.
/ Fehler bei wiederholtem ein- und ausschalten des manuellen Notstrom-schalters behoben.
/ Abschaltgeschwindigkeit wurde erhöht für den Hard Limit Mode bei der dynamischen Einspeisebegrenzung.
/ State 1196 Fehlauslösungen behoben - Slave Geräte welche in der ”Einspeisebegrenzung - Mehrere Wechselrichter limitieren”
Funktion gesteuert werden können nicht mehr in diesen Fehlerzustand versetzt werden.
/ Avahi durchstarten stündlich”

closed 2025-03-20 by redpomodoro