redpomodoro/fronius_modbus #24: Feature: Add support for export limitation

rodxavier, 2025-03-04:
As discussed in this [thread](https://community.home-assistant.io/t/automate-fronius-soft-limit/465124/33), it would be good if you can add support for controlling/setting registers 40232 and 40236 to control soft limiting exports. This is particularly helpful where feed-in prices can be zero/negative such as the case for Amber(AU).

RikMe, 2025-03-30:
Second that! Any chance we could have these registers available to read and write to? Would make curtailment dead easy.

Great thread with a manual solution here: https://community.home-assistant.io/t/automate-fronius-soft-limit/465124/58?page=4

Would be great to be able to access these parameters trough your integration

40232 = Rate limit from 10000 (100%) to 100 (1%) the lower number the lower output
40234 = Timeout in seconds for the output limit to expire (if HA fails to set this for any reason)
40236 = Enable / disable ratelimit 1=Enable 0=Disable.

smuliv, 2025-05-10:
Yep - this would be very useful for me too to not have to pay for hourly negative prices!

ManfredGroh, 2025-05-10:
That would be absolutely great!!!

oldman943, 2025-05-11:
I agree - this would be amazing indeed!

RikMe, 2025-05-11:
@redpomodoro is this somewhere on your roadmap or horizon? Would be great to have it, certainly on days with negative prices like we have these days.

If not I will have to look into a more manual configuration solution.

Tacmatricx, 2025-08-22:
+1

I got this integration up and running this week and this is my only issue. Have to set Export Limitation above Modbus control until this is resolved (Also on Amber AU and need to limit PV export during the day when FIT is negative).

Love your work!

rodxavier, 2025-08-24:
@bjarnebuchmann, actually exactly the same. Your instructions seems clearer too. 😄