redpomodoro/fronius_modbus #25: Error after upgrade to HA 2025.3.0

vistalba, 2025-03-06:
After updating to HA 2025.3.0 today I run into this error when I try to change e.g. PV charge limit:

`Die Aktion number/set_value konnte nicht ausgeführt werden. object of type 'int' has no len()`

soylentOrange, 2025-03-08:
Same for me. Fresh install of the addon didn't solve this problem.

In my home assistant log, I see this warning:
2025-03-08 07:13:45.621 WARNING (MainThread) [py.warnings] /config/custom_components/fronius_modbus/number.py:89: RuntimeWarning: coroutine 'Hub.toggle_busy.<locals>.wrapper' was never awaited
  self._hub.set_grid_charge_power(value)

vistalba, 2025-03-08:
@soylentOrange a downgrad of the addon to 0.1.14 solved the issue for me. Does this work for you as well?

soylentOrange, 2025-03-08:
@vistalba Thanks for the hint! Tried 0.1.15 and it worked already.

redpomodoro, 2025-03-18:
Closing as this should be resolved. Please reopen if still any issue with v0.1.16 or higher.

closed 2025-03-18 by redpomodoro

soylentOrange, 2025-03-18:
@redpomodoro The problem of not being able to set the limits only occurs in v0.1.16. The earlier versions (v0.1.15 in my case) seem to work fine.

soylentOrange, 2025-03-18:
@redpomodoro The problem of not being able to set the limits only occurs in v0.1.16. The earlier versions (v0.1.15 in my case) seem to work fine.

Can't reopen the issue (lacking permissions)