redpomodoro/fronius_modbus #26: Secondary Generator

soylentOrange, 2025-03-07:
I added a second power meter to my fronius system (Symo GEN24 10.0 Plus) as a secondary generator. 
Until recently the added power of the inverter and of the secondary generator were reported correctly using fronius_modbus, but after updating my inverter to the most current firmware (ROW 1.35.4-1) and also updating fronius_modbus to v0.1.6 only the inverters power is being reported (via the sensor.fronius_inverter_symo_gen24_10_0_plus_load).

In the fronius web-ui, solar-net and so on the summed values are still being correctly reported.

Is there any way to get the old behaviour of fronius_modbus back?

soylentOrange, 2025-03-08:
No worries, a fresh install of the addon fixed the problem. Might be related to upgrading the inverter's firmware...

soylentOrange, 2025-03-08:
I was too happy too soon, unfortunately it doesn't work as expected. The stated power is still only the consumption of the PV power of the inverter and not the aggregated power. 

The value reported by the solarnet integration is correct. Just fronius modbus is giving me an off value.

Tested with 0.1.16 and 0.1.15.

redpomodoro, 2025-03-20:
Just so I understand the situation correctly. You have 2 power meters 1 at grid connection and 1 at the secondary generator (non Fronius I assume).

And the load (power consumption) is incorrect, because it doesn't include power generated by the secondary generator?

soylentOrange, 2025-03-20:
Exactly. 
I've integrated a sunspec compliant meter into the fronius inverter as a secondary generator that is visible in the web frontentend of the inverter (screenshot here: https://github.com/soylentOrange/OpenDTU/blob/master/docs/screeshot_solarweb.png). The values reported by solarnet and the inverter's web frontend sum up the solar production of the inverter and this secondary generator. A full description of the power meter and it's integration is given here: https://github.com/soylentOrange/OpenDTU

Until I updated Home Assistant and the inverter's firmware, your fronius_modbus integration showed the same behaviour (summing up both pv power) but afterwards only the power of the inverter is being reported. The production of the secondary generator is not being reported from fronius_modbus.

I'm not sure if such a setup is in widespread use, but I'm struggling with the inconsistency between solarnet and fronius_modbus. Right now, I'm using fronius_modbus (v0.1.15) in order to control the battery modes, eversthing else is being handled by the solarnet integration - yet, they don't really align well.

psyinf, 2025-12-16:
Unfortunately the integration doesn't allow for more power-meters to be added. The values reported are correct however as the primary inverter doesn't know about other generators. Those are used for the calculation of the values used in the solarweb view however. To my knowledge Fronius doesn't allow you to access those calculated values