redpomodoro/fronius_modbus #28: Freeze of HA after upgrade to v0.1.6

vistalba, 2025-03-17:
I now have tried many things by myself but I didn't nailed it down yet.
When I update to the latest version 0.1.6 of the Fronius integration (and also upgrade byd_battery_modbus to 0.1.15) my homeassistant is feezeing every few seconds. It is almost unusable. While this happens CPU is runnig at 97-100% forever.
I then have to disable both integrations and downgrade it over HACS to the previous version.

What I've tried so far:

- Modified manifest & hub.py to use old pymodbus version. Checked installed version inside "homeassistant" docker with `pip show pymodbus`
- Disabled only the BYD battery integration
- Fronius v0.1.6 and downgraded BYD to v.0.1.14.

When downgrading both integrations (Fronius = 0.1.5 / BYD = 0.1.14) AND reboot the whole VM it is working fine. I do really not have a clue what happens here :(

Anyone else having such issues?

vistalba, 2025-03-18:
I don't want. The problem is, that the sensors stopping updating when on v0.1.6 and v0.1.15 after some time.
So I tried out if a rollback to the pymodbus version of v0.1.5/v0.1.14 would solve the problem. Basicly to rule out that this leds to the issue. But that wasn't successful.

redpomodoro, 2025-03-18:
And nothing showing up in the logs? Just that all sensors stop updating?