redpomodoro/fronius_modbus #29: Noob question about HA integrations

Jaywatts, 2025-03-18:
Hi, sorry for this question. I'm sure there will be a very strait forward answer but I'm very new to HA.
I installed Virtualbox on my PC and installed HA. I have downloaded the Fronius Modbus and added as an integration.
As seen in the screenshot attached I can only find "Configuration" and "Diagnostic" to add to my Dashboard.
Can someone let me know what I'm doing wrong please? 

[image]

redpomodoro, 2025-03-20:
What happens if you settings -> devices & services -> devices and then filter by Fronius Modbus Integration? You should see then the devices that got configured after you installed the integration

RikMe, 2025-04-11:
https://github.com/redpomodoro/fronius_modbus/issues/24

closed 2025-05-20 by Jaywatts