redpomodoro/fronius_modbus #44: Integration Crashing with HA Version  2025.7.0

DanielStockinger, 2025-07-03:
Hi Guys,

after installing the latest HA Core Version 2025.7.0, the integration is crashing.
Before it was working fine for me.

[image]

Logger: homeassistant.config_entries
Source: config_entries.py:3733
First occurred: 6:20:23 AM (2 occurrences)
Last logged: 6:21:15 AM

Error occurred loading flow for integration fronius_modbus: cannot import name 'unpack_bitstring' from 'pymodbus.utilities' (/usr/local/lib/python3.13/site-packages/pymodbus/utilities.py)

[image]

Any help is appreciated.

Many Thanks
Daniel

max-t-d, 2025-07-05:
Quick feedback from my side: my integration loads fine on 2025.7.1

andriuskr, 2025-07-07:
2025.7.1 getting same error

`Logger: homeassistant.setup
Source: setup.py:343
First occurred: 21:58:32 (1 occurrence)
Last logged: 21:58:32

Setup failed for custom integration 'fronius_modbus': Unable to import component: cannot import name 'unpack_bitstring' from 'pymodbus.utilities' (/usr/local/lib/python3.13/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 998, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1058, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/fronius_modbus/__init__.py", line 18, in <module>
    from . import hub
  File "/config/custom_components/fronius_modbus/hub.py", line 13, in <module>
    from .froniusmodbusclient import FroniusModbusClient
  File "/config/custom_components/fronius_modbus/froniusmodbusclient.py", line 8, in <module>
    from .extmodbusclient import ExtModbusClient
  File "/config/custom_components/fronius_modbus/extmodbusclient.py", line 13, in <module>
    from pymodbus.utilities import unpack_bitstring
ImportError: cannot import name 'unpack_bitstring' from 'pymodbus.utilities' (/usr/local/lib/python3.13/site-packages/pymodbus/utilities.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 343, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1018, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1010, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1058, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/fronius_modbus/__init__.py", line 18, in <module>
    from . import hub
  File "/config/custom_components/fronius_modbus/hub.py", line 13, in <module>
    from .froniusmodbusclient import FroniusModbusClient
  File "/config/custom_components/fronius_modbus/froniusmodbusclient.py", line 8, in <module>
    from .extmodbusclient import ExtModbusClient
  File "/config/custom_components/fronius_modbus/extmodbusclient.py", line 13, in <module>
    from pymodbus.utilities import unpack_bitstring
ImportError: cannot import name 'unpack_bitstring' from 'pymodbus.utilities' (/usr/local/lib/python3.13/site-packages/pymodbus/utilities.py)`

petikehc, 2025-07-11:
@DanielStockinger Created a fix for it under #45, if you don't want to wait for it to be merged, you can also update your files locally.