redpomodoro/fronius_modbus #48: Numbers values

WojtusBLO, 2025-07-22:
Can setup correct values. Look at screen shot:

[image]

[image]

DAVIZINH0, 2025-08-09:
hello.
I have the same problem.
With i select the "pv charge and discharge limit" i cand push the values i want.
in my case my batery have a capacity of 11.059W
I want pv charge limit in 90%, so its 9.953W
and discharge limite in 10% so its 1.106W

But i cant set this values manualy and cant set it with a "action" :-(

any idea how to solve this?

DAVIZINH0, 2025-09-07:
And now Im Trying to config the grid charger power to 2kw.
But if I out the number 2000 it changes automatically to other number that i dont understand.
What is the problem with this numbers??