redpomodoro/fronius_modbus #51: Integration not start after HA 2025.8.2

DAVIZINH0, 2025-08-15:
I updated today to 2025.8.2
And updated pyton scripts pro to de latest version.
And the fronius modbus stop working.
I dont konw if the problem was ha update or python

The error is this:

[image]

jetzlstorfer, 2025-08-15:
Same issue on my end after upgrading.

epenet, 2025-08-15:
Custom integrations should follow the "core" version and be adjusted accordingly

DAVIZINH0, 2025-08-15:
@redpomodoro sorry sayi your name. but all the people using the integration and upgrade to 2025.8.1 will be affected :-(

Toxo666, 2025-08-16:
have the bug 2.

here is a hotfix for manual implementation:

`diff --git a/custom_components/fronius_modbus/manifest.json b/custom_components/fronius_modbus/manifest.json
index e3a2c7c..b6d42ab 100644
--- a/custom_components/fronius_modbus/manifest.json
+++ b/custom_components/fronius_modbus/manifest.json
@@
-    "requirements": ["pymodbus==3.8.3"],
+    "requirements": [],
diff --git a/custom_components/fronius_modbus/extmodbusclient.py b/custom_components/fronius_modbus/extmodbusclient.py
index 9a8c0d1..12ab345 100644
--- a/custom_components/fronius_modbus/extmodbusclient.py
+++ b/custom_components/fronius_modbus/extmodbusclient.py
@@
-from pymodbus.utilities import unpack_bitstring
+from pymodbus.pdu.pdu import unpack_bitstring
`

MickeSB, 2025-08-16:
Issue #44 is the same and the solution petikehc made works fine. Download his files and replace the old ones, restart HA. 
Solution in pull request #45 https://github.com/redpomodoro/fronius_modbus/pull/45

Atleast it worked perfect for me

oriziena, 2025-08-16:
I also applied it manually and can confirm that it works with those small patches. thanks

DAVIZINH0, 2025-08-16:
> Issue [#44](https://github.com/redpomodoro/fronius_modbus/issues/44) is the same and the solution petikehc made works fine. Download his files and replace the old ones, restart HA. Solution in pull [...]

Thanks, works for me!