redpomodoro/fronius_modbus #53: Integration not start after HA 2025.9.0

matake31, 2025-09-04:
It won't load after update to HA 2025.9.0.

This from my log.

2025-09-04 09:09:48.141 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Fronius for fronius_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 751, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/fronius_modbus/__init__.py", line 45, in async_setup_entry
    await entry.runtime_data.init_data()
  File "/config/custom_components/fronius_modbus/hub.py", line 55, in wrapper
    raise error
  File "/config/custom_components/fronius_modbus/hub.py", line 49, in wrapper
    result = await func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/fronius_modbus/hub.py", line 61, in init_data
    await self._hass.async_add_executor_job(self.check_pymodbus_version)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/fronius_modbus/hub.py", line 73, in check_pymodbus_version
    raise Exception(f"pymodbus {version('pymodbus')} found, please update to {self.PYMODBUS_VERSION} or higher")
Exception: pymodbus 3.11.1 found, please update to 3.8.3 or higher

andreastomincssi, 2025-09-04:
same issue, just one day after using it :(

waltshomeassistant, 2025-09-04:
Same here.  Subscribed. 

Hopefully can find a fix soon, one of the handiest integrations there is!  Well done on all the great work so far.

Rainman67, 2025-09-04:
Please can somebody tell me, what exactly to do? I replaced the files "manifest.json" and "extmodbusclient.py" but after a restart of home assistant the problem persists.

DAVIZINH0, 2025-09-04:
hello. 
same problem. can someone tell us how to change to fix, until apears a new version?
thanks a lot!!!

rodxavier, 2025-09-04:
@Rainman67, if you're replacing manually, you also need to do `hub.py`.

@DAVIZINH0, you can replace the files manually, or you can replace the installed version with my fork until the changes are merged to the main repo.

rodxavier, 2025-09-05:
@DAVIZINH0, are you replacing it with files from here, https://github.com/redpomodoro/fronius_modbus/pull/49/files?

4webat, 2025-09-05:
https://github.com/redpomodoro/fronius_modbus/pull/49/files  
Thanks, all files updated = works

PDelvaux, 2025-09-06:
It works after replacing the files and the restarting HA

SuperMasterPhoenix, 2025-09-06:
If you do not want to replace all files from the mentioned PR, but only hub.py, extmodusclient.py and manifest.json you also have to modify hub.py as this file contains some code which also requires the rest of the PR:
Delete lines 201-206 and 292-302 in hub.py. That's it—after an HA restart, it should work again.

heasti, 2025-09-07:
@rodxavier Replacing your files did not work for me. I received the following message:
Traceback (most recent call last): File "/config/custom_components/fronius_modbus/hub.py", line 203, in async_refresh_modbus_data update_result = await self._client.read_export_limit_data() ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ AttributeError: 'FroniusModbusClient' object has no attribute 'read_export_limit_data'

I then removed lines 203-207. in hub.py Now it works again.

lutzvahl, 2025-09-10:
Replacing all the files manually from PR #49 works for me. Can we please get this one merges asap that any new update will not break things again? Thanks