redpomodoro/fronius_modbus #55: IDEA: adding more maintainers to the repo

jetzlstorfer, 2025-09-10:
First of all, I really like this integration! Thanks @redpomodoro for contributing it to the community!

To make it even better and help with ongoing development, I'd like to suggest adding more people as maintainers to this repo. As we can see, e.g. in this discussion #53 this repo has a quite active community. 

I thought maybe other people that are enjoying this HACS integration and are knowledgeable of coding might want to step up and join this repo to help further development and maintenance? 
As we can see from issue #53 it currently takes some time for this repo to be updated and release a new version. Maybe if others can help, the work on the current maintainer can be taken off his/her shoulder and distributed to more folks from the community?

So anyone who would be willing to contribute could just comment on this issue? 
Of course, @redpomodoro has to be fine with it :)

Just an idea :)

flixhsw, 2025-09-10:
I absolutely agree with that proposal and would be willing to help.

The extension is one of the most important for my smart home, as all of my "energy control" automations rely on it.
And I feel a bit frustrated, that after 4 weeks my PR https://github.com/redpomodoro/fronius_modbus/pull/50 is still open without a single comment.
So I would be really willing to help to improve the quality and maintenance of this extension. So that it can finally be an official homeassistant extension :-)

DAVIZINH0, 2025-09-24:
someone can talk with the owner about this idea?

DAVIZINH0, 2025-09-25:
@redpomodoro are you still here???

maybe a fork its posible if the developer not continue updating?

im not a coder, but i thing we have some good ideas to improve the integration, and now there are some critical things to fix:
- 2 inverters in one installation
- the max %charge of battery not works.
- the max power to charge not works

larsxschneider, 2025-09-29:
Maintaining an open source component for Home Assistant can be both time-consuming and _very_ stressful. I completely understand if @redpomodoro chooses not to continue as the maintainer.

Is anyone interested in taking over maintenance of a fork? Unfortunately, I don't have the capacity to commit to this task myself right now.

DAVIZINH0, 2025-10-01:
I think the better solution is adding more maintainers to this repo. 
The work of redpomodoro is great but is clear than he not have time to maintain all the posible improves.
Now there are some interesting ideas of improve this integration like parallel inverters.
Or problems like the max charge rate thant must be solved.

A fork is a posible solution, but maybe not the best.

and something important. if someone open a fork, how many people that have the coder skill will help??
because if only one person have this skill, maybe the problem will be the same in a sort period of time

callifo, 2025-11-14:
I thought as this thread had time to run and didnt get an outcome, I decide to spin up my own fork here

https://github.com/callifo/fronius_modbus

I have already committed all the pr's and changes I could find. Put credit for those authors in the change log. 

Happy to take new PRs going forward and add shared additional contributors if there is interest.